GuidoJeuken-6512/lambda_heat_pumps #15: Fehler mit Intergration

UweRau, 2025-08-17:
### Description

Hallo,
hatte die Lämbda Wärmepumpenintegration 21. Juni über hacs installiert. Diese lief bis gestern. Seit gestern bekomme ich keine Entitäten mehr angezeigt. habe die integration deinstalliert und wieder neu installiert. jetzt bekomme ich beim Integration hinzufügen folgende Meldung "Der Konfigurationsfluss konnte nicht geladen werden: 500 Internal Server Error Server got itself in trouble"

Installationsmethode Home Assistant OS
Core 2025.8.2
Supervisor 2025.08.1
Operating System 16.1
Frontend 20250811.0

### Reproduction Steps

Version der Intergration vor dem Fehler kann ich nicht sagen, aktuell installierte Version ist die 1.09.
Die alte Konfiguration habe ich heute deinstalliert und danach versucht die neue über HACS aufzuspielen.

### Log

2025-08-17 19:45:55.572 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration lambda_heat_pumps which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-08-17 20:35:05.421 ERROR (MainThread) [aiohttp.server] Error handling request from 192.168.178.177
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 41, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 83, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 181, in post
    return await self._post_impl(request, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 188, in _post_impl
    return await super()._post_impl(request, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 83, in _post_impl
    result = await self._flow_mgr.async_init(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1446, in async_init
    flow, result = await self._async_init(flow_id, handler, context, data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1482, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1704, in async_create_flow
    handler = await _async_get_flow_handler(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.hass, handler_key, self._hass_config
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 3779, in _async_get_flow_handler
    await _load_integration(hass, domain, hass_config)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 3756, in _load_integration
    await async_process_deps_reqs(hass, hass_config, integration)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 584, in async_process_deps_reqs
    await requirements.async_get_integration_with_requirements(
        hass, integration.domain
    )
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 54, in async_get_integration_with_requirements
    return await manager.async_get_integration_with_requirements(domain)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 170, in async_get_integration_with_requirements
    await future
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 159, in async_get_integration_with_requirements
    await self._async_process_integration(integration, done)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 182, in _async_process_integration
    await self.async_process_requirements(
        integration.domain, integration.requirements
    )
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 264, in async_process_requirements
    self._raise_for_failed_requirements(name, missing)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 292, in _raise_for_failed_requirements
    raise RequirementsNotFound(integration, [req])
homeassistant.requirements.RequirementsNotFound: Requirements for lambda_heat_pumps not found: ['pymodbus==3.6.1'].

### Version

2025.08.1

### Home Assistant Core Version

2025.8.2

### Home Assistant Operating System Version

16.1

GuidoJeuken-6512, 2025-08-28:
sind jetzt alle Fehler behoben? Dann würde ich das Issue gern schließen.

michaelkoslowski, 2025-08-28:
Hi.Ja es ist alles ok. Läuft super.Viele Grüße Michael