GuidoJeuken-6512/lambda_heat_pumps #29: Runtime-Error

MichaSU99, 2025-09-20:
### Description

Zunächst einmal herzlichen Dank für diese hervorragende Integration und die ebenso erstklassige Dokumentation dazu!
Meine Lambda ist von Zewotherm, wobei die Steuerung wohl die Original-Steuerung von Lambda ist (kein Zewo-Branding).

Einen Fehler kann ich auf der Benutzeroberfläche nicht erkennen, benutze aktuell aber nur ein paar Sensoren.

Im HA-Log taucht aber untenstehender Runtime-Error auf:

Schon mal Danke und sorry, falls dieser Fehler bereits bekannt ist.

### Reproduction Steps

Der Fehler taucht täglich einmal auf...

### Log

2025-09-20 00:00:00.418 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback _TrackPointUTCTime(hass=<HomeAssistant RUNNING>, job=<Job time change listener 0:0:0 <function setup_cycling_automations.<locals>.reset_daily_sensors at 0x7f9298885da0> HassJobType.Callback <bound method _TrackUTCTimeChange._pattern_time_change_listener of _TrackUTCTimeChange(hass=<HomeAssistant RUNNING>, time_match_expression=([0], [0], [0]), microsecond=417487, local=True, job=<Job track time change 0:0:0 local=True HassJobType.Callback <function setup_cycling_automations.<locals>.reset_daily_sensors at 0x7f9298885da0>>, listener_job_name='time change listener 0:0:0 <function setup_cycling_automations.<locals>.reset_daily_sensors at 0x7f9298885da0>', _pattern_time_change_listener_job=<Job time change listener 0:0:0 <function setup_cycling_automations.<locals>.reset_daily_sensors at 0x7f9298885da0> HassJobType.Callback <bound method _TrackUTCTimeChange._pattern_time_change_listener of ...>>, _cancel_callback=<bound method _TrackPointUTCTime.async_cancel of _TrackPointUTCTime(hass=<HomeAssistant RUNNING>, job=<Job time change listener 0:0:0 <function setup_cycling_automations.<locals>.reset_daily_sensors at 0x7f9298885da0> HassJobType.Callback <bound method _TrackUTCTimeChange._pattern_time_change_listener of ...>>, utc_point_in_time=datetime.datetime(2025, 9, 20, 22, 0, 0, 417487, tzinfo=datetime.timezone.utc), expected_fire_timestamp=1758405600.417487, _cancel_callback=<TimerHandle when=1010609.0027654171 ...()>)>)>>, utc_point_in_time=datetime.datetime(2025, 9, 19, 22, 0, 0, 417487, tzinfo=datetime.timezone.utc), expected_fire_timestamp=1758319200.417487, _cancel_callback=<TimerHandle when=924209.0027661324 ...()>)() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1534, in __call__
    self.hass.async_run_hass_job(self.job, self.utc_point_in_time)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 937, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1847, in _pattern_time_change_listener
    hass.async_run_hass_job(self.job, localized_now, background=True)
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 937, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/config/custom_components/lambda_heat_pumps/automations.py", line 33, in reset_daily_sensors
    _update_yesterday_sensors(hass, entry_id)
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/config/custom_components/lambda_heat_pumps/automations.py", line 127, in _update_yesterday_sensors
    yesterday_entity.set_cycling_value(daily_value)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/config/custom_components/lambda_heat_pumps/sensor.py", line 916, in set_cycling_value
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_verify_state_writable()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 987, in _async_verify_state_writable
    raise RuntimeError(f"Attribute hass is None for {self}")
RuntimeError: Attribute hass is None for <entity unknown.unknown=unknown>

### Version

1.3.0

### Home Assistant Core Version

2025.9.4

### Home Assistant Operating System Version

16.2

GuidoJeuken-6512, 2025-11-02:
Das neue Release ist veröffentlicht, der Fehler sollte jetzt verschwunden sein.
Freue mich auf ein Feedback

MichaSU99, 2025-11-03:
Hallo, nach dem Update scheint dieser Fehler nicht mehr aufzutauchen. Dafür aber einige Andere, die vorher nicht da waren... Ich prüfe, ob die bereits als Issue eingestellt wurden...
Auf jeden Fall vielen Dank für Ihr Engagement!