GuidoJeuken-6512/lambda_heat_pumps #31: Feature Request: Extend Modbus Control Functions (Buffer, Heating Circuits)

Manuel2030, 2025-09-27:
### Is your feature request related to a problem?

First of all, thank you for this great integration!

Currently, only boiler control is exposed. It would be very useful to also add heating circuit controls via Modbus registers, so that setpoints and operating modes can be adjusted directly from Home Assistant.

Requested Entities: Heating circuit (ModulNr. 1-12)

-  Operating mode(optional) also by boiler suggest
-  heating temp

This would allow users to adjust heating circuit settings without going into the heat pump’s own menus, and integrate them more seamlessly with PV surplus and smart home automation.

Thanks again for your work on this project!
If I can help let me know! :) 

### Describe the solution you'd like

Feature request idea

### Describe alternatives you've considered

added functions

rreinhardd, 2025-10-06:
Indeed this would be a great feature to have. I use another MODBUS Integration to do this, but I would prefer to have everything in one integration.

I think the most importent Elements to set from HA would be:

- ✅ Offset for the Boiler target Temp
- ❌ Offset for the Flow Line Temp of the heating circout
- ❌ Set the Operating State of HP1
- ❌ Set the Operating State of HC*
- ❌ Set the Operating State of Boiler
- ❌ Set the Operating State of Buffer

[image]

rreinhardd, 2025-12-14:
First, thanks for the great work with the component. Regarding the Climate Cotrols. I think Climate Controls are more intuitive from the UI prospective in many use cases. Lets say, I have guest and want to turn the heat up or down by 2 degree. Or I am out for a few days and want to turn it down by 4°. Or the Boiler Temp: I have a thermal Solar Power plant and the next days look sunny so I want to turn the Water temp to 40°. Or I know I want a bath tonight and turn up the water to 55. Yes I could make Scripts or Scenes for that - but I think a Slider is the most intuitive way of controlling themperature.

Also one comment on the minimum Boiler temp - could it be that you limit this to 40° in the your integration while Lambda allows 25° ? This would be another of my use cases: Since I cannot change the „Betriebsart“ from the Boiler (to off) I want to set the temperature to a minimum, and leave the boiler heating to my thermal solar plant.

magor77, 2025-12-15:
Hi @all
I think this would be a very helpful addidtion.
We have 2 HC and a fireplace in one of the appartements. If I make a fire I usally connect the the HP Control and change the operating state because I don't need any more heat on this appartement for the next couple of hours. This would be helpful to do in HA. 
An similar Feature in my old heating was the "Party Mode", in this case you push a Button on the room control and the heating is lowered to night mode until midnight.

GuidoJeuken-6512, 2025-12-15:
> ```
> service: modbus.write_register
> data:
>   hub: EU13L          # Name deines Modbus-Hubs
>   slave: 1            
>   address: 5006   # oder 5106, 5206, 5306
>   value:
>     - "{{ (3 | int) | bitwise_and(65535) }}"
> ```
@MTTPoll 
Wenn ich versuche auf 5006 zu schreibe, ändert sich nichts in der Lambda, HC Betriebsart bleibt unverändert.  "{{ (3 | int) | bitwise_and(65535) }}" ist ja = 3 also Auto-Heat. 
Habe auch 2 = Automatik ausprobiert, in meiner Lambda ändern sich leider nichts.

[image]

rreinhardd, 2025-12-16:
Im Video von @MTTPoll ist zu sehen, dass sich zwar der Zustand des HK ändert, aber der die "Betriebsart" bleibt unverändert. Daher kann es vorkommen, dass der sensor.eu13l_hc1_operating_mode_2 nicht den tatsächlichen Zustand des HK widerspiegelt nach einer Aktualisierung via Modbus. Daher der "Schattensensor" und die damit verbundene Problematik, die @GuidoJeuken-6512 schildert.

GuidoJeuken-6512, 2025-12-23:
@magor77 

Ohne ein konfiguriertes Raumthermometer sollten die Einstellungen so aussehen, natürlich mit deinen Werten:

[image]

Wenn eine Raumthermomenter aktiviert wurde in der Lambda und in der Integration sind weitere Werte einstellbar:

[image]

GuidoJeuken-6512, 2025-12-24:
ich kann es nachstellen, behebe den Fehler im nächsten Release